COAST-Lab/c3po #25: Confirm sensor sleep commands

SUPScientist, 2024-09-27:
Current sensor sleep commands:

https://github.com/COAST-Lab/ConductivitySensor_Old/blob/42f6ad3f61daf9208db27fdebbddd982a49db844/Firmware/AtlasConductivity_20240724/src/AtlasConductivity_20240724.cpp#L191C3-L192C67

Doubtful this works as intended. Confirm with (1) datasheet and (2) power monitoring during usage. Much likelier to be a call to the `EC` and `RTD` objects than a text-based I2C command.

gracelower, 2025-02-04:
@efarq added some sleep commands / delays on 20 December 2025, which are now reflected in the .cpp code file; these delays allowed the Boron and EZO circuits to interact cooperatively

SUPScientist, 2025-02-04:
Note that even if those commands work as described/intended (questionable), 0.5 mA is still too high for sleep current, so we'll need to add a solution to further reduce power draw, such as a FET switch (like in TinyCamML).

gracelower, 2025-02-04:
Good to know, thanks for clarification. Will definitely investigate this / keep an eye on it as we move forward.

efarq, 2025-07-30:
I think with the updated source code I made in commit #ea04c91 , this is no longer an issue, as I do not use the EC.send_cmd("sleep") function at all. However, I will still try to use a multimeter to assess the power draw during the different stages to ensure.