COAST-Lab/c3po #26: Check sig figs in data string

SUPScientist, 2024-09-27:
In 

https://github.com/COAST-Lab/ConductivitySensor_Old/blob/42f6ad3f61daf9208db27fdebbddd982a49db844/Firmware/AtlasConductivity_20240724/src/AtlasConductivity_20240724.cpp#L153-L156

it appears that different numbers of digits are reported from the temp vs. cond sensor. Check to see how many the sensor reports and record/transmit all.

gracelower, 2024-10-07:
Line 153 snprintf(data, sizeof(data), "%li,%.5f,%.02f",
Line 154 real_time, // if it takes a while to connect, this time could be offset from sensor recording
Line 155 temp,
Line 156 cond

format specifiers `%.5f and %.02f` control how many digits are shown after the decimal point for floating-point values (temp and cond)
- .5f = five digits after decimal point
- .02f = two digits after decimal point

––> Adjust specifiers based on what sensors report so we receive same number of digits for cond + temp sensors

gracelower, 2024-11-01:
Attempting to update code / fix this issue, but returning this error when trying to install SdFat library: "Access token provided is invalid"

https://api.particle.io/v1/libraries/SdFat?version=latest

Is SdFat no longer available? Or is my software not compatible / do I need to update something?

SUPScientist, 2024-11-01:
That's a particle login issue most likely. Check that you're logged into your particle account in VS Code

efarq, 2025-07-30:
[image]
[image]

The current format for the temp/conductivity data points is 0.2f, or 2 places after the decimal. I've attached two images from the publishing spreadsheet that show that during low conductivity readings, it reports values to the hundredths place; however, during high conductivity readings, it does not report any values to in the hundredths place. 

Additionally, it sounds like we could go to 0.3f for the RTD sensor, but I don't necessarily think we need to? And the 0.2f seems sufficient for the conductivity, but I would be glad to hear any one else's opinion.

SUPScientist, 2025-07-31:
Thanks for following up on this. 

It appears that cond is reporting 4 sigfigs, so I think the 0.2f is fine (though it's nowhere close to accurate to a single µS/cm, so if it reports, for example, 12.51 µS/cm, we wouldn't trust it at better than ~ 10).

For temp, if it reports 3 decimal places, might as well save 'em and we can figure out how to post-process with that level of resolution after the fact. It doesn't cost us anything since we're nowhere close to max publish message length.