COAST-Lab/c3po #29: UART --> I2C

gracelower, 2024-10-31:
Have spent much time attempting to switch EZO EC and RTD circuits from UART to I2C mode based on the documentation provided, but:
- RTD will only blink cyan/green (UART mode)
- EC won't flash any colors
https://www.whiteboxes.ch/docs/tentacle/t2-mkII/#/protocols

SUPScientist, 2024-11-01:
I don't think this will fix your problem with the Atlas EZO circuit, but to at least eliminate one variable, I recommend flashing a blink LED script to your Boron with cell disabled. E.g., this one: https://github.com/SUPScientist/Smart-Coasts/blob/main/Class-01-Intro/Blink_LED.cpp. This might resolve the issue @efarq was describing re: Particle wanting you to claim the Boron, which I don't think should be necessary yet as long as you use `SYSTEM_MODE(SEMI_AUTOMATIC);`.

SUPScientist, 2024-11-01:
Next, could you confirm which documentation you are following to put the device into I2C mode? Is it https://files.atlas-scientific.com/EZO_RTD_Datasheet.pdf?

gracelower, 2024-11-01:
> Next, could you confirm which documentation you are following to put the device into I2C mode? Is it https://files.atlas-scientific.com/EZO_RTD_Datasheet.pdf?

Yes, that is the protocol I have been using, also described here: https://www.whiteboxes.ch/docs/tentacle/t2-mkII/#/protocols

I just worked through these steps again and seem to still only receive cyan/green blinking on the RTD. The EC switched to I2C properly (dark blue).

SUPScientist, 2024-11-01:
My questions are probably unhelpful because I know you're really thorough, but since I can't exactly see what pins you have connected to which, can you confirm that this is exactly what you're doing:
[image]

Idea 2: use the Boron's 3V3 as your power to the VCC port on the EZO board.

Idea 3: if that still doesn't work, you could also try to send a command via UART (hook up the Boron's 3V3 and GND to the RTD EZO's VCC and GND and then the Boron's Rx to the EZO's Tx and vice versa). Then put a line in your `void loop()` function that is, I think, `Serial1.println('I2C,100');` Note that it will then use 100 as the address if it works. You also have to reboot it (disconnect/reconnect power to EZO) to see if new color appears to indicate I2C mode.

gracelower, 2024-11-01:
Update:
Was able to get RTD to cyan/aqua blinking --> indicates I2C communication
EC is still showing solid blue --> indicates UART connection
When powering from Boron's 3V3 instead of VUSB, I am able to get a solid green, which I think indicates standby mode, though does not indicate whether it's in I2C or UART.
Will try Idea 3 soon! Thanks for the assistance!

gracelower, 2024-11-07:
[image]
[image]
[image]

closed 2024-11-07 by gracelower