COAST-Lab/c3po #31: Ezo_i2c libraries "nonexistent"

gracelower, 2024-11-11:
[image]

[image]

Ezo_i2c, Ezo_i2c.h, other variations of this library "do not exist" when I try to install them; above are screenshots from compiling code without library
I am logged in through my Particle account, so that's likely not the issue (?)
- That was the issue for the SdFat library

Potential causes:
- I am incorrectly trying to install a library
- I have an incorrect setting in my file pathway

SUPScientist, 2024-11-12:
Weird. Must be a workflow issue or similar. I just cloned the entire repo, opened the c3po/Firmware/AtlasConductivity_20240724 folder in VS Code/Workbench, compiled without touching anything, and it worked fine. 

A few thoughts...  
- The libraries for the ezo stuff are already in /lib, so you don't need to manually install anything. 
- Are you opening only the Particle project (c3po/Firmware/AtlasConductivity_20240724) in VS Code or the whole repo (should be the former)? 
- Anything else will probably take more troubleshooting. I might be able to do a quick zoom tomorrow (Tues) if you want to give that a shot and share your screen to show me what you're seeing. I'm at NC Coastal Conf Wed and then back Thu.

gracelower, 2024-11-12:
I figured it was likely a pathway issue on my end. Unfortunately, my schedule is pretty packed today, but I'll investigate this deeper tomorrow morning. Thanks!

gracelower, 2024-11-15:
I removed my cloned repo and started over by cloning the c3po repo again and opening the whole repo in VS Code, which eliminated the pathway error of accessing the Ezo library. However, I'm now receiving errors claiming that a couple of files actually within the i2c library can't be accessed, so I'm going to reboot this process to try again. I think my issues are all stemming from pathway problems, so I'm going to approach cloning very deliberately.

gracelower, 2024-11-15:
[image]

These are the errors I'm receiving, even after I started from scratch with a new cloned repo. It appears that the files `I2Cdev.h` and `MPU6050.h` can't be accessed or don't exist, though they are called by
`#include "I2Cdev.h" #include "MPU6050.h"` at the beginning of the src file.

SUPScientist, 2024-11-15:
This looks like the problem described in email by @BentleySettin: "One VS issue I ran into before was opening the entire GH repo and not just the designated project folder in VS. Opening the entire repo led to many issues such as with libraries."

The solution is to open only the Particle project (c3po/Firmware/AtlasConductivity_20240724) in VS Code. That is, navigate to AtlasConductivity_20240724 when opening the project, not the whole repo. The left-hand side shows that you have the whole repo open.

gracelower, 2024-11-15:
I may have found my issue -- I had tried what @BentleySettin suggested (opening solely the Particle project) and I received the same error, so I retraced my steps & realized I had been manually setting the device & OS to "Boron" & "4.2.0" like we did in the summer for the water level sensor. I then tried setting to "Boron" and "6.1.1" which was the given OS when opening just the Particle project (and was then given the ability to "clean" the "dirty" file) and compiled successfully. Are we using OS 6.1.1 for these sensors moving forward?

SUPScientist, 2024-11-15:
Glad it works! I'm DeviceOS agnostic—whatever works is fine for now. But sticking with something relatively up to date like 6.1.1 does sound wise. When you open at the Particle Project level, it should automatically know what version had been used because that comes along in one of the settings files (I always forget which). But it sounds like this is solved, so nice work! I'll let you close if that's correct.

closed 2024-11-15 by gracelower