COAST-Lab/c3po #32: Test code w/ sensors

gracelower, 2024-11-28:


gracelower, 2024-12-13:
Tried with only EZO EC circuit, Atlas Isolated Carrier Board, EZO EC probe, and Boron / Adalogger.
Wired VCC / 3V3, GND / GND, SDA / SDA, SCL / SCL.
Still receiving blank serial monitor.

EZO EC is correctly showing dark blue LED which is I2C.
Boron blinks green, which on Particle website, is indicated as "connecting to network." Could this be a source of error? I would think the device would still be recording internally on the microSD card even if no network connection exists, though Dr. Hino said we do not have a microSD card in Chapel Hill, so I haven't been able to check if it has been storing data.

gracelower, 2024-12-16:
The Boron has been claimed via the Particle app (special thanks to @efarq for help with that!)
I will go back to `PUBLISHING 0` now and see if that makes any adjustments to the LED status!

gracelower, 2024-12-16:
Updates:
- I have tried `PUBLISHING 0` and unfortunately am only getting the device to blink green; I will read Particle forum posts to see if anyone in the "Particle-verse" has experienced this issue 
- Maybe I can manually turn off the network connection through the Particle Console / app?
- Found a converter at my house; no files on microSD card (expected, since sensor is being blocked)

SUPScientist, 2024-12-16:
Have you gone through the getting started steps that you used (and for that matter helped write!) for the open-water-level sensor (https://github.com/COAST-Lab/Open-Water-Level/blob/main/Firmware/Getting_Started%20with%20Cell-Enabled%20Water%20Level%20Sensing.md)? For instance, have you confirmed that the Boron is running the same Device OS that you are compiling for (it looks from the following to be 6.1.1):
https://github.com/COAST-Lab/c3po/blob/ce125387664c0b1ae50e006cd57f2a87b746afdf/Firmware/AtlasConductivity_20240724/.vscode/settings.json#L8

My next advice would be to take a giant step back and confirm that you can get the Boron to do anything that you want it to do, such as blink an LED. If you can strip away any unnecessary complexity to see what is and isn't working and then build it back up, that will help identify the source of the problem(s).

gracelower, 2024-12-19:
In reference to: https://github.com/COAST-Lab/c3po/issues/34#issuecomment-2555770926

[image]

gracelower, 2024-12-20:
Pictured is an example of output after ...
- wiring Boron / Adalogger to Port 4 on Tentacle T3 Shield instead of upper rows of pins
- reformatting microSD card
- adding delays to code to account for sleep timing & serial connections

[image]

closed 2024-12-20 by gracelower