COAST-Lab/c3po #34: (Re)gain control of Boron

SUPScientist, 2024-12-17:
### Problem
As described in https://github.com/COAST-Lab/c3po/issues/32, the Boron does not seem to be responding to any flashed code. Instead, regardless of whether cell is on or off, the status LED continues to blink green. @gracelower has confirmed that it has been device restored to Device OS 6.1.1 and that code is being compiled on Device OS 6.1.1. Workbench successfully compiles code and Boron appears to indicate successful flashing, but then Boron goes back to blinking green and does not execute flashed code.

### Recommended troubleshooting
Try the device doctor: https://docs.particle.io/tools/doctor/. I haven't used this yet myself but it is recommended in several threads pertaining to issues with Borons incessantly blinking green. 

@efarq tagging you to notify you of this new issue.

SUPScientist, 2024-12-17:
OK, I don't know the source of the problem yet, but I uncovered a major clue when trying to reproduce @gracelower's issue. I reproduced it simply by cloning this repo, compiling, and flashing the firmware onto a new Boron using Device OS 6.1.1 (as has been used in this repo. I did a bunch of stuff to try to ascertain what was wrong and, without getting into all those details, what I learned was that I had to go back to the Open-Water-Level's Device OS version (4.2.0) to get the Boron to respond as expected.

This suggests that somewhere between Device OS 4.2.0 and 6.1.1, Particle changed something in either `SYSTEM_MODE(SEMI_AUTOMATIC)`, `Cellular.off()`, and/or `Particle.connect()` most likely. I'll keep digging.

One apparent workaround to test the sensors would be to keep `#define PUBLISHING 0` and do a device restore USB to 4.2.0 and switch the Device OS version to 4.2.0 in VS Code/Workbench. Not an elegant final solution but worth testing.

SUPScientist, 2024-12-17:
Trying to learn more here: https://community.particle.io/t/new-ish-behavior-with-system-mode-semi-automatic-and-manual/68799.

gracelower, 2024-12-18:
Aha! Great catch! I also just replicated the Blink_LED process using 4.2.0 instead of 6.1.1 and was able to get the Boron to breathe a white LED instead of blink green.
Working on flashing the cond/temp code to the device now and seeing if any results appear in the serial monitor.

SUPScientist, 2024-12-18:
@efarq and @gracelower please read the community thread that I linked three posts ago in this issue before Friday (and, if possible, test on your system, Grace). Specifically that last part about running `particle update` in the Particle CLI which appears to help download/install the missing dependency. I'm confused (as I noted in the thread) as to why the Device Restore USB tool doesn't do this, but at least we know more about how these devices work now. 

I can now confirm that my Boron behaves as expected running Device OS 6.1.1 and the current firmware (https://github.com/COAST-Lab/c3po/commit/76d3c5f8d662edbfc362859a7f6788e4f9936ec2) so I'm closing this issue and we can return to troubleshooting the sensors themselves.

closed 2024-12-18 by SUPScientist

gracelower, 2024-12-19:
I am confirming that by using the `particle update` and `particle serial inspect` commands as detailed in your forum discussion, I am able to successfully compile and flash Blink_LED code to the Boron using OS 6.1.1, resulting in it breathing white instead of breathing green. I am also able to successfully flash the AtlasCond code to the Boron under OS 6.1.1, but then lose the LED altogether. I have deduced that this is not a physical power / connection issue because I am still able to see LED signals when interacting with the Mode and Reset buttons (on the Boron), so I am wondering if this may have to do with sleep commands; perhaps the Boron is going to sleep after flashing due to commands in the AtlasCond code? For reference, I have `PUBLISHING = 0` and have not changed the `SECONDS_BETWEEN_MEASUREMENTS` from 360. This might be worth investigating in person tomorrow (Friday 12/19).
https://docs.particle.io/troubleshooting/led/

SUPScientist, 2024-12-19:
Wonderful! Thanks for the update. I think we can fully move out of this thread now as this issue seems to be resolved.

Yes, I'd put lots of $$$ on this being simply a matter of sleep timing. If you have the serial monitor open when you run the code (and have it configured to automatically reopen when connection is re-established), you should see it print to the monitor that it is going to sleep. I saw the same behavior yesterday. I believe it's doing exactly what it should be!