COAST-Lab/c3po #36: Lower body housing

gracelower, 2024-12-17:


gracelower, 2025-01-23:
I had been considering a 3D-printed cap for the PVC pipe that would have either a:
- larger hole for cable gland, through which 4-wire cable is threaded & therefore removable if needed
- smaller hole through which 4-wire cable is threaded & epoxied

I had also considered a purchasable flat-top pipe cap (example [here](https://www.mcmaster.com/2389K104/)) to which the 3D-printed chassis could be epoxied ... not sure if purchasing has any benefits over 3D printing this same design with the chassis already attached --> Should pipe cap lay flush with 3" nominal PVC, or cover it like a cap?

Could also use a rubber stopper + cable entry through PVC pipe itself, though would have to epoxy, not cable gland due to curvature of PVC pipe

SUPScientist, 2025-01-24:
@gracelower @efarq check out https://github.com/COAST-Lab/Open-Water-Level/blob/main/Fabrication/PVC-Enclosure/Enclosure-and-Assembly/PVC-End-Cap/PVCTop.stl for an example of a custom end-cap designed for sealing off one end of a 1.5" SCH 40 PVC pipe with a hole for a cable gland. It is designed so that the smaller diameter of this end-cap fits inside the PVC, then the lid sits on the top edge of the PVC. We'll likely add a little extra sealant after final assembly for additional peace of mind.

gracelower, 2025-02-04:
Notes from Bentley –

O-ring for water level:
https://www.mcmaster.com/9464K291/

Resources for O-ring designs:
https://ceetak.com/o-ring-calculator/
https://dichtomatik.com.mx/catalogos/diseno-de-alojamientos.pdf
https://www.globaloring.com/o-ring-groove-design/

Math for water level O-ring

[image]

gracelower, 2025-03-28:
@SUPScientist Thanks for the heads up. Will definitely keep that in mind as we proceed!

.stl files for 3D designs have been added to the repo –
Commit: 97b56c0644a478266c53f73f97005d2dce2971e3

gracelower, 2025-04-02:
Purchase order sent for:
- O-rings https://www.mcmaster.com/9464K734/
- Cable glands https://www.polycase.com/cg3-cable-glands#CG-32
- 3" nominal ID sch 40 PVC pipe
- Mounting screws for chassis

gracelower, 2025-04-21:
Notes from build days 4/17-18:

- 3D printed pipe caps are too big; need to decrease thickness of all aspects of .stl files for both upper and lower pipe caps (actual dimensions pictured below; measurements taken in person on 4/18)
- O-ring calculations did not provide us with a plausible O-ring design, so we found a better O-ring solution in the machine shop. need to find origin of new O-rings we were given so we can use those dimensions moving forward
- need to increase diameter of temp probe entry hole in .stl file (possibly was a printer issue though)
- need to source hose clamps for a future comprehensive DIY kit

gracelower, 2025-04-26:
PVC housing i3 Interlink & probes, attached to pole with hose clamps
Note: 3D-printed caps here had to be modified heavily post-print to accomodate O-rings and cable glands; thinner print designs are in progress

[image]

gracelower, 2025-05-05:
[image]

gracelower, 2025-05-08:
[BottomCap_NoGrooves_20250508.stl](https://github.com/gracelower/c3po/blob/main/Hardware/BottomCap_NoGrooves_20250508.stl)
[TopCap_NoGrooves_20250508.stl](https://github.com/gracelower/c3po/blob/main/Hardware/TopCap_NoGrooves_20250508.stl)