COAST-Lab/c3po #38: Consider other circuitry for Atlas probes

SUPScientist, 2025-01-30:
### Problem
The interlink/tentacle shield is relatively large, forcing us to use a larger lower body housing, as described in https://github.com/COAST-Lab/c3po/issues/36. 

### Potential Solution
The Atlas individual EZO (isolation) boards might suffice for our needs, e.g., the one used here: https://atlas-scientific.com/kits/conductivity-k-1-0-kit/ (isolated) and here: https://atlas-scientific.com/kits/sma-temperature-kit/ (non-isolated). Note that we wouldn't need new probes (temp or cond) nor EZO circuits, just the carrier boards. I believe that these carrier boards can use I2C just like the interlink/tentacle shield. One key difference is that we would need two different boards in this proposed scenario, but assuming they use I2C, the same 4 wires coming to the lower body housing from the upper body housing could be spliced/split off to those two carrier boards via separate wires. (This is effectively what the interlink/tentacle board does anyway, but via traces on a PCB rather than wires.) 

The advantage here is huge space savings. Much easier to fit two of those carrier boards with their EZO circuits inside a little lower body housing. 

@gracelower @efarq please review, consider this option, and provide feedback when possible. @gracelower, I'm not recommending a change in course yet so please proceed with your lower body housing designs.

efarq, 2025-01-31:
I think this definitely could be a good option so we can use a smaller diameter PVC in the future. I'm also pretty sure we already have those little carrier boards - I can check. 

One question I have - what's the difference in the "isolated" verses "non-isolated" ?

gracelower, 2025-02-04:
Hi all,

I have an EC and RTD board here in Chapel Hill and could definitely play around with a design using those at some point. As @SUPScientist mentioned, I can stick with the i3 Interlink for now so we can get a working prototype, but I imagine it's definitely doable to use the two separate boards in the future. My only question would be how this would change our wires -- would we use two 4-wire waterproof cables? But would these then have to connect to two Borons in theory (obviously not ideal)? Or perhaps there is a way to combine the wiring from the two boards still into a single 4-wire cable and Boron. Not sure if my explanation makes any sense, but something to consider if we do end up switching from the i3 Interlink for the sake of size.

SUPScientist, 2025-02-04:
Yes, splicing the wires ought to work, so no need for additional wires.

Please put this thread on the backburner and proceed with what you have been working on (i3 interlink/tentacle shield version) to maintain momentum there.