COAST-Lab/c3po #50: Configure correct security settings for Sheet

SUPScientist, 2025-05-22:
### Challenge
Current google sheet may have too open of permissions. I'm admittedly unsure. @efarq created the Sheet, I believe, and would likely have control over permissions.

### Recommendation
Configure sheet to not allow any arbitrary person to edit it. Allow permissions for small core team (@efarq, @gracelower, @SUPScientist, @BentleySettin would likely suffice) as well as Particle Cloud (which needs special read/write access for the integration).

It's possible that it already has correct or best available permissions, but it is critical that we confirm this.

gracelower, 2025-05-22:
Excellent note. On my end, it appears that the spreadsheet was "editable" by anyone on the internet (as pictured below). This can easily be fixed by adjusting the sharing settings to "Restricted," which allows only people added to the sheet (i.e. individuals added to the NASA Google Folder / Conductivity folder). I can do this, or I can let @efarq handle it in case adjusting the settings messes with any of the data collection (I don't think it should, but just in case).
[image]

SUPScientist, 2025-05-23:
> in case adjusting the settings messes with any of the data collection

My concern is that in the instructions we follow to configure the Particle/Sheets connection, it says, "Who has access --> Anyone" and I'm not sure what the ramifications of changing that might be. I suppose it should be an easy test, though maybe best done on a test Sheet first to see if/how it works rather than messing with the Sheet that is our only operational data access currently.